w3c/automotive #410: Proposal to add current value to the response on subscribe request

crea7or, 2021-09-11:
It's useful because allow us to avoid additional requests since sometimes you need to know current value before subscription will be triggered.

```
{
  "action": "subscribe",
  "data": {
    "dp": {
      "ts": "2020-04-15T13:37:00Z",
      "value": "2000"
    },
    "path": "Vehicle/Drivetrain/InternalCombustionEngine/RPM"
  },
  "requestId": "6578",
  "subscriptionId": "12345",
  "ts": "2020-04-15T13:37:00Z"
}
```

wonsuk73, 2021-09-14:
?@crea7or, Thanks for the proposal. Can I ask specific use cases for this?

crea7or, 2021-09-14:
@wonsuk73 a health signal for example. App want to know a health of some component and subsribes to the health signal. But what is current health of this component is unknown until app will request signal explicitly as separate request or wait untill subscribtion will be trigered.

wonsuk73, 2021-09-14:
@crea7or, I am not sure for this kind of use case we need to add this feature because we need to focus major features.

crea7or, 2021-09-15:
@wonsuk73 this is only a small optimization so absolutely low priority.

wonsuk73, 2021-09-15:
@crea7or, Thanks.

closed 2021-09-15 by wonsuk73