w3c/automotive #417: Multi set request as performance optimization

crea7or, 2021-09-14:
Current version of set request allow us to set only one value, even if we'll use filter operation. However it will be very useful to have multi set request. Example - CAN frames processing. CAN frame may contain few signals and we should send them only-by-one right now. Wich is not optimal since all the network code is called as many times as we have signals. It would be much more optimal if we'll fill one reqeust for a few signals at once. 
Additionally this multi set reqeust can be used for `agregates` (if spec finally will have it).

wonsuk73, 2021-09-14:
@crea7or, for better understanding of your idea do you have any proposal?

erikbosch, 2021-09-15:
@crea7or - are you looking for something like below, a possibility to use "post" on branch level and instead of just giving a value give a path/value array as argument. so that you e.g. could set both PerformanceMode and GearChangeMode in the same request

```
 POST /Vehicle/Drivetrain/Transmission   HTTP/1.1
            Host:127.0.0.1:1337
            Authorization:Bearer eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJuYW1lIjoiSm9obiBEb2UifQ.xuEv8qrfXu424LZk8bVgr9MQJUIrp1rHcPyZw_KSsds
            {
              // Path relatives to /Vehicle/Drivetrain/Transmission (or full?)
              “data”:[{“path”:”PerformanceMode”, "value": "sport"},{“path”:”GearChangeMode”, "value": "manual"}]
            }
```

crea7or, 2021-09-15:
@erikbosch yep, but prefer full paths like this, because in theory signals from the one frame can be in the different roots:

```
{
  "action": "set",
  "data": [
    {
      "path": "Vehicle/Drivetrain/Transmission/PerformanceMode",
      "value": "sport"
    },
    {
      "path": "Vehicle/Drivetrain/Transmission/GearChangeMode",
      "value": "manual"
    }
  ],
  "requestId": "5687"
}
```

crea7or, 2021-09-15:
If one set command will be in priority, one value set can be just and object instead of array of the objects:

```
{
  "action": "set",
  "data": 
    {
      "path": "Vehicle/Drivetrain/Transmission/PerformanceMode",
      "value": "sport"
    },
  "requestId": "5687"
}
```

crea7or, 2021-09-27:
Another idea is to specify root node(branch) like this:
```
{
  "action": "set",
  "data": [
    {
      "path": "PerformanceMode",
      "value": "sport"
    },
    {
      "path": "Gear",
      "value": "2"
    }
  ],
  "requestId": "5687",
  "root": "Vehicle/Drivetrain/Transmission"
}
```

This syntax will help us to subscribe to the branches to work with them as with composite types.

closed 2021-11-19 by UlfBj