w3c/automotive #447: Publication cycle of ths specs in WD level

wonsuk73, 2022-02-17:
Recently I made three contributions to apply cross references between VISSv2 Core and VISSv2 Transport. It's helpful to readers of this specs when they check the parts of the specs related with contents they see. But to overcome link breaking issue between recent editor's drafts(Core and Transport) URLs of officially published docs are used because it use fixed url regardless of published version the specs (e.g. https://www.w3.org/TR/viss2-core/). 

When I tried to make links between docs there is the issues because of the big gap between published spec and editor's draft. As you know we had published FPWD of Core and Transport specs around 8 months ago, so there are many changes in the specs.

So I would like to propose to make a  publication cycle of the specs in working draft level as every quarter or every half of the year.

UlfBj, 2022-03-01:
Carine will set it up so that the published version will be the same as the branch.

wonsuk73, 2022-03-02:
@UlfBj Great! Thanks!  I think there are some more works we need to do after published WD.

closed 2022-03-08 by UlfBj