w3c/automotive #450: Inverse range filtering?

UlfBj, 2022-03-08:
The range filtering currently supports 
unbounded range: defined by the "logic-op" and "boundary" parameters 
bounded range: two sets of "logic-op" and "boundary" parameters, combined by an AND operator. 

The bounded range support can define something like 0 < X < 5, but not the inverse of it, i. e. "all X not inside the bounded range". 
To support this, the mentioned AND operator needs to be replaced by an OR operator. 
This could be supported by adding an optional parameter, "combination-op", that have either the value "AND" or "OR". This parameter MUST only be used in a bounded range expression, and only be present in the first of the two boundary expressions. 
Examples:
{"boundary-op":"gt", "boundary": "0",  "combination-op": "AND"}, {"boundary-op":"lt", "boundary": "5"}  // 0<X<5
{"boundary-op":"lte", "boundary": "0",  "combination-op": "OR"}, {"boundary-op":"gte", "boundary": "5"} // inverse of above

In the example the key name "logic-op" is changed to "boundary-op". A similar change should be done in the change filtering syntax, "logic-op" changed to "diff-op"

tguild, 2022-03-08:
Discussed on today's call
https://www.w3.org/2022/03/08-auto-minutes.html#t02

erikbosch, 2022-04-05:
An alternative approach could be to have it on type level based on this example from transport:

```
 {
                "action": "subscribe",
                "path": "Vehicle/Drivetrain/FuelSystem/Level",
                "filter": {"type":"range", "value":[{"logic-op":"gt", "boundary":"49"}, {"logic-op":"lt", "boundary":"51"}]},
                "authorization": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJuYW1...Zw_KSsds",
                "requestId": "6578"
              }
```

And for filter possible allow arbitrary complex statements like:
```
 "filter": {"type":"or", "filter":[
    {"type":"range", "value":[{"logic-op":"gt", "boundary":"17"}]},
    {"type":"and", "filter":[
         {"type":"range", "value":[{"logic-op":"gt", "boundary":"14"}]},
         {"type":"range", "value":[{"logic-op":"le", "boundary":"11"}]},
       ]}
   ]}
```

UlfBj, 2022-04-07:
See PR 452.

closed 2022-04-07 by UlfBj