w3c/automotive #462: documentation: subscription timestamp

aw-muc, 2022-11-30:
Within the https://raw.githack.com/w3c/automotive/gh-pages/spec/VISSv2_Transport.html#subscribe documentation, the subscriptionNotification is described as followed

Object Name | Attribute | Type | Required
-- | -- | -- | --
action | Action | Yes
subscriptionId | string | Yes
data | object/array | Yes

Within the examples, the following is documented:
Notification:

              {
                "action": "subscription",
                "subscriptionId": "12345",
                “data”: {“path”: ”Vehicle.Drivetrain.FuelSystem.Level”, 
                        “dp”: {“value”: ”50”, “ts”: ”2020-04-15T13:37:00Z”}
                },
                **"ts": "2020-04-15T13:37:00Z"**
              }
               
As you can see the example has an additional timestamp mentioned, which is not part of the documentation of the subscribeNotification;

Can you please clarify, what is the correct implementation?

UlfBj, 2023-01-02:
Subscription notification messages shall have a timestamp. 
That is added to the table in PR#471.

aw-muc, 2023-01-09:
Awesome; @UlfBj will the issue automatically be closed, when pull request is merged?

UlfBj, 2023-01-09:
@aw-muc It will be closed at the WG meeting when it is merged. Thanks for finding this documentation error.

UlfBj, 2023-01-20:
Fixed in PR#471

closed 2023-01-20 by UlfBj