w3c/automotive #463: Subscription handling on error or JWT auth issues (timed out)

aw-muc, 2022-11-30:
I did not find any clear instruction about the handling of subscriptions if they run into an error like authentication error. 

For example, i have a time based subscription, where the servers sends every 5 seconds notification about the current value I have subscribed with an valid JWT for time X; after time X the internal check runs into an issue, because the JWT is not longer valid; From the documentation the client now gets every 5 seconds an error notification that the JWT is outdated because of auth errors.

Is this a valid behavior, or are there any valid reasons for the severs to close a subscription if he recognizes an error that can't be fixed by the server e.g. JWT is outdated.

UlfBj, 2023-01-02:
It is in my view probably a reasonable behavior that a server closes a subscription when it recognizes an error that it cannot fix, and sends an error message to the subscriber.

aw-muc, 2023-01-09:
Would it make sense that these type of behavior are documented somewhere; Because at the moment I havn't found any kind of documentation pointing into that direction;

UlfBj, 2023-01-09:
@aw-muc It will be documented in the specification, but a discussion and a decision must first take place in the WG. 
Thanks for bringing this issue up.

UlfBj, 2023-04-02:
Fixed in PR478

closed 2023-04-02 by UlfBj