w3c/automotive #465: Add in-line privacy and security considerations to VISS transport

samuelweiler, 2022-12-01:
https://www.w3.org/TR/viss2-transport/ has no in-line Privacy nor Security writeup.  I suggest the WG to prioritize the architectural stuff in https://github.com/w3c/automotive/issues/464 over this issue.

tguild, 2023-04-04:
Transport cannot be used without Core, inclined to refer to its privacy and security considerations. Additional security considers exist for each underlying protocol, what to reference or direct implementers to, OWASP as WoT did?

https://www.w3.org/TR/2023/CR-wot-discovery-20230119/#security-considerations

provide link to COVESA Mira board of some potential architectures - there is no one defined one - including some where VISS may be used.

W3C Automotive Best Practices is stalled but still desirable, intent to do with COVESA Data Expert Group.

https://www.w3.org/TR/2019/NOTE-wot-security-20191106/

UlfBj, 2023-10-03:
Architecture figures:

Slide 5:
https://docs.google.com/presentation/d/1y7xJabLK9iUubxR8_4z1M6rlci_fT13V/edit#slide=id.p14

 

Autosar Data Arch Diagram draft:
https://drive.google.com/drive/folders/10P5KanrU-CwR4xfQnL0rPxAorlPuuYWD