w3c/automotive #467: VISS Core: why is access control non-normative?

samuelweiler, 2022-12-06:
https://www.w3.org/TR/viss2-core/#access-control-model seems pretty code.  Why is it non-normative?

erikbosch, 2022-12-07:
The main argument during the review was that the use of the Access Grant/Token servers may not be feasible for all deployments. That there could be deployments where for example grants are pre-provisioned in factory or by FOTA updates rather than being dynamically fetched as in the picture. So the decision at that time was that VISS shall show a possible way to get an access token, but it shall not make it mandatory to use that methodology, i.e. it shall for example be possible for a vehicle OEM to choose a different technology.

UlfBj, 2023-04-18:
Fixed in PR479

closed 2023-04-18 by UlfBj