w3c/automotive #473: VISS Core&transport - unclear MAY with enumeration

Ilmarinen100, 2023-01-18:
The VISSv2 Core and transport spec contain multiple definitions like this:
> subscriptionId - string - Value returned by the server to uniquely identify each subscription. The value MAY be an integer or a Universally Unique Identifier (UUID).

This appears to say that the id can alternatively be chosen arbitrarily as long as it is encoded into a string
Is that the intended meaning or was a more strict interpretation planned, e.g. the value MUST be either of the following:
* an integer (range?) encoded as a decimal string
* a uuid encoded as lowercase hexadecimal string (with standard 8-4-4-4-12 notation)

UlfBj, 2023-02-01:
>>This appears to say that the id can alternatively be chosen arbitrarily as long as it is encoded into a string

Yes, that is the correct interpretation of the somewhat unclear statement in the spec. This statement will be updated to become clearer.

UlfBj, 2023-03-31:
Fixed in PR475

closed 2023-03-31 by UlfBj